karawin/Ka-Radio32 #252: Internal DAC output doesn't work on Ka-Radio32 V2.1 R3

Ant-12, 2022-01-26:
To solve this problem, these [changes in audio_renderer.c](https://github.com/karawin/Ka-Radio32/commit/bbfc792299456b6daf6e680cab61b74aa0a8d77a#diff-2d6ea1d293433b656577bc4bc0a59f6c0fc73369f497ba7a8bcdb7305a770084) must be reverted.

What was the reason of these changes?

P.S.
I also got a [similar error](https://github.com/espressif/esp-idf/issues/3013), trying to flash firmware using "make flash".
The solution for me was to change "Bootloader config  ---> Bootloader log verbosity" to "Info".

karawin, 2022-01-31:
I think it comes from the spdif test. I will change it.

Ant-12, 2022-02-03:
Release 2.2, Revision 1  - still getting noise

Ant-12, 2022-02-05:
Release 2.2R1 and prerelease(2021-02-05) - built in DAC still doesn't work...

There are too many "2.2R1" versions. Maybe you will add a timestamp of build for prereleases?

karawin, 2022-02-05:
Release 2.2R1 before today was a mistake. 
I cannot test dac. 
I added the missing parts but .... 
Please help.

Ant-12, 2022-02-05:
You did not revert **all** changes.
There need to correct [this line](https://github.com/karawin/Ka-Radio32/blob/85448f56de2ba000638a3979ea41e89812a39f3c/components/audio_renderer/audio_renderer.c#L555).

And please decrease "Bootloader log verbosity".

P.S.
Have you tried to build Ka-radio32 under linux OS? I got better results in decoding of HE-AACv2 streams(64 kbit) using this:
Toolchain version: crosstool-ng-1.22.0-97-gc752ad5
Compiler version: 5.2.0
ESP-IDF v3.3.5-dirty

karawin, 2022-02-05:
Right, I never thought that it may be the cause.
It works now.
"And please decrease "Bootloader log verbosity"."
Ok

I will reload a good R1 this night.

KaRadio is under linux, but KaRadio32 is under windows.
In fact I dont have so much time to change, and a bit lazy ;-)

I will migrate to esp_idf v5 as soon as possible but ....
Thanks

karawin, 2022-02-05:
Generated  in release mode, now 
 fdkaac_decoder: pcm_size 8192, channels: 2, sample rate: 48000, object type: 2, bitrate: 62812

Works well.
Updated and ready for OTA.

Ant-12, 2022-02-06:
Now I can confirm that built in DAC works.
But your builds still have problems(stuttering) with playing HE-AACv2 streams(64 kbit) through S/PDIF output...

closed 2022-02-06 by Ant-12