karawin/Ka-Radio32 #253: Upgraded new version 2.2 with problems

tototek, 2022-01-28:
Before upgrading to 2.2 Rev 0, the stations that have a bitrate of 128k mp3 sounded good. Now they are heard "as in chunks" they are not heard continuously and smoothly as it worked in the previous version. 
It happens to me with almost all the stations that I listened to . 
Radio stations with .aac that were previously unheard now work fine.
You may be able to upgrade to earlier versions from the settings page?.

Regard
[audio_2022-01-28_15-21-43.zip](https://github.com/karawin/Ka-Radio32/files/7960948/audio_2022-01-28_15-21-43.zip)
s

karawin, 2022-01-31:
Try to upgrade one more time please.

mars16, 2022-02-01:
Hello! After updating the firmware version from 2.1.r2 to  to 2.2.R0-2.2.R1, there have been problems with multiple threads.
1 https://stream.gtf.club:1250/stream began to interrupt, the buffer is filled with 0%
2 http://www.5-tv.ru:80/radio.mp3 stopped playing
3 https://stream.pcradio.ru:443/mdsst_en_2-hi     is constantly restarting 
4 http://stream.radiojar.com:80/hcrb063nn3quv   began to play more stable and increased buffer occupancy 
Situation is checked on two devices KaRadio32: ESP32wroom+ VS1053B + 4MB PSRAM
It will be great if these problems manage to eliminate. Thanks for your work!

karawin, 2022-02-01:
### https://stream.gtf.club:1250/stream
May be a problem due to the lack of RAM, but it play.
Try to find an http url.

## On wroom esp32 and vs1053:
##CLI.URLSET#: https://stream.gtf.club
##CLI.PATHSET#: /stream
##CLI.PORTSET#: 1250
##CLI.OVOLSET#: 0
I (12397465) webclient: IP: 6691ab3e   ADDR:62.171.145.102
##CLI.VOL#: 62
I (12400075) webclient: Set Song buffer to 25k
##CLI.ICY0#: Ð¡Ð¾Ð²ÐµÑ‚ÑÐºÐ°Ñ ÑÑÑ‚Ñ€Ð°Ð´Ð°
##CLI.ICY3#: http:\/\/disco-radio.ru\/
##CLI.ICY4#: Retro, 80's, 70's
##CLI.ICY5#: 256
##CLI.ICY6#: Ð¡Ð¾Ð²ÐµÑ‚ÑÐºÐ°Ñ ÑÑÑ‚Ñ€Ð°Ð´Ð° Ð²Ñ€ÐµÐ¼ÐµÐ½ Ð½Ð°ÑˆÐµÐ¹ Ð¼Ð¾Ð»Ð¾Ð´Ð¾ÑÑ‚Ð¸
##CLI.ICY7#:  bitrate=256
##CLI.META#: Ð’Ð˜Ð \"ÐÑ€Ð¸ÑÐ»ÑŒ\" â€“ Ð’ÑÐµ Ñ€Ð°Ð²Ð½Ð¾ Ñ‚ÐµÐ±Ñ Ð»ÑŽÐ±Ð»ÑŽ Ñ (1980)
I (12400725) audio_player: Buffer fill 81%, 20898 // 25600 bytes
W (12400725) renderer: chip revision 1, enabling APLL
##CLI.PLAYING#

I (12408245) audio_player: Buffer fill 6%, 1671 // 25600 bytes
I (12414605) audio_player: Buffer fill 50%, 12993 // 25600 bytes
I (12421315) audio_player: Buffer fill 70%, 18009 // 25600 bytes
I (12428695) audio_player: Buffer fill 21%, 5470 // 25600 bytes
I (12435055) audio_player: Buffer fill 67%, 17173 // 25600 bytes
##CLI.META#: Ð®Ñ€Ð¸Ð¹ ÐÐ½Ñ‚Ð¾Ð½Ð¾Ð² â€“ ÐœÐ¾Ñ€Ðµ (1982)
I (12442445) audio_player: Buffer fill 27%, 7142 // 25600 bytes
I (12449195) audio_player: Buffer fill 22%, 5851 // 25600 bytes
I (12455575) audio_player: Buffer fill 54%, 13829 // 25600 bytes
I (12461935) audio_player: Buffer fill 96%, 24696 // 25600 bytes

## On wrover and i2s:
No problem

##CLI.URLSET#: https://stream.gtf.club
##CLI.PATHSET#: /stream
##CLI.PORTSET#: 1250
##CLI.OVOLSET#: 0
I (7600863) webclient: IP: 6691ab3e   ADDR:62.171.145.102
##CLI.VOL#: 165
I (7604083) webclient: Set Song buffer to 400k
##CLI.ICY0#: Ð¡Ð¾Ð²ÐµÑ‚ÑÐºÐ°Ñ ÑÑÑ‚Ñ€Ð°Ð´Ð°
##CLI.ICY3#: http:\/\/disco-radio.ru\/
##CLI.ICY4#: Retro, 80's, 70's
##CLI.ICY5#: 256
##CLI.ICY6#: Ð¡Ð¾Ð²ÐµÑ‚ÑÐºÐ°Ñ ÑÑÑ‚Ñ€Ð°Ð´Ð° Ð²Ñ€ÐµÐ¼ÐµÐ½ Ð½Ð°ÑˆÐµÐ¹ Ð¼Ð¾Ð»Ð¾Ð´Ð¾ÑÑ‚Ð¸
##CLI.ICY7#:  bitrate=256
##CLI.META#: ÐÐ»Ð»Ð° Ð˜Ð¾ÑˆÐ¿Ðµ Ð¸ Ð¡Ñ‚Ð°Ñ…Ð°Ð½ Ð Ð°Ñ…Ð¸Ð¼Ð¾Ð² â€“ Ð¯ Ð²Ð°Ñ Ð½Ðµ Ñ‚Ð¾Ñ€Ð¾Ð¿Ð»ÑŽ (1976)
##CLI.PLAYING#

I (7605693) audio_player: Buffer fill 13%, 56006 // 409600 bytes
I (7605973) audio_player: Buffer fill 16%, 66037 // 409600 bytes
W (7605973) renderer: chip revision 1, enabling APLL
I (7611353) audio_player: Buffer fill 23%, 96585 // 409600 bytes
I (7617053) audio_player: Buffer fill 28%, 116647 // 409600 bytes
I (7623343) audio_player: Buffer fill 29%, 119155 // 409600 bytes
I (7629783) audio_player: Buffer fill 28%, 117262 // 409600 bytes
I (7636033) audio_player: Buffer fill 29%, 120827 // 409600 bytes

karawin, 2022-02-01:
### http://www.5-tv.ru:80/radio.mp3
## On wroom
No problem
##CLI.URLSET#: http://www.5-tv.ru
##CLI.PATHSET#: /radio.mp3
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
I (13094355) webclient: IP: a50ec97   ADDR:151.236.80.10
##CLI.VOL#: 62
I (13094405) webclient: Set Song buffer to 40k
I (13094505) webclient: Header: Moved
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: https://www.5-tv.ru
##CLI.PATHSET#: /radio.mp3
##CLI.PORTSET#: 443
I (13095655) webclient: watermark : 24c  588
I (13095655) webclient: IP: a50ec97   ADDR:151.236.80.10
I (13097785) webclient: Set Song buffer to 25k
##CLI.ICY0#:  DomRadio
##CLI.ICY3#:  \/
##CLI.ICY5#:  128
##CLI.ICY7#:  samplerate=44100;channels=2;bitrate=128
##CLI.META#: 
I (13098875) audio_player: Buffer fill 82%, 21010 // 25600 bytes
W (13098875) renderer: chip revision 1, enabling APLL
##CLI.PLAYING#

I (13119195) audio_player: Buffer fill 53%, 13809 // 25600 bytes
I (13138855) audio_player: Buffer fill 53%, 13632 // 25600 bytes

## On wrover
Problem no data. I have to increment the waiting time. It seems that the stream is very slow to come.
But sometime it works

##CLI.URLSET#: http://www.5-tv.ru
##CLI.PATHSET#: /radio.mp3
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 0
I (299761) webclient: IP: a50ec97   ADDR:151.236.80.10
##CLI.VOL#: 165
I (300001) webclient: Header: Moved
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: https://www.5-tv.ru
##CLI.PATHSET#: /radio.mp3
##CLI.PORTSET#: 443
I (301161) webclient: watermark : 1c4  452
I (301161) webclient: IP: a50ec97   ADDR:151.236.80.10
I (303301) webclient: Set Song buffer to 400k
##CLI.ICY0#:  DomRadio
##CLI.ICY3#:  \/
##CLI.ICY5#:  128
##CLI.ICY7#:  samplerate=44100;channels=2;bitrate=128
##CLI.META#: 
##CLI.PLAYING#

I (305121) audio_player: Buffer fill 16%, 66873 // 409600 bytes
W (305121) renderer: chip revision 1, enabling APLL
I (323851) audio_player: Buffer fill 20%, 82250 // 409600 bytes
I (343611) audio_player: Buffer fill 20%, 84069 // 409600 bytes

karawin, 2022-02-01:
### https://stream.pcradio.ru:443/mdsst_en_2-hi

On both cpu, not found.
Tested on a browser: The file you requested could not be found
Wrong url?

mars16, 2022-02-01:
#INFO:""#
I (234428) vs1053: watermark: 504  1284
I (234818) webclient: SSL Cleanup 1 Socket: 60
I (234948) webclient: watermark : 300  768
##CLI.NAMESET#: 5 MDS-2
##CLI.URLSET#: https://stream.pcradio.ru
##CLI.PATHSET#: /mdsst_ru_2-hi
##CLI.PORTSET#: 443
##CLI.OVOLSET#: -20
I (234958) webclient: IP: a714d85f   ADDR:95.216.20.167
##CLI.VOL#: 197
W (235658) wolfssl_mp: exceeds max bit length(2048)
W (235668) wolfssl_mp: exceeds max bit length(2048)
W (235728) wolfssl_mp: exceeds max bit length(2048)
I (239078) webclient: Set Song buffer to 400k
I (239078) webserver: playstationInt: 5, g_device: 5
I (239478) webclient: Header: Moved
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: https://str4.pcradio.ru
##CLI.PATHSET#: /mdsst_ru_2-hi
##CLI.PORTSET#: 443
I (240508) webclient: SSL Cleanup 1 Socket: 63
I (240628) webclient: watermark : 300  768
I (240628) webclient: IP: 3c14d85f   ADDR:95.216.20.60
W (241348) wolfssl_mp: exceeds max bit length(2048)
W (241368) wolfssl_mp: exceeds max bit length(2048)
W (241428) wolfssl_mp: exceeds max bit length(2048)
I (244708) webclient: Set Song buffer to 400k
##CLI.ICY0#:  mdsst_ru_2
##CLI.ICY3#:  http:\/\/mdsst.ru\/
##CLI.ICY5#:  64
##CLI.ICY6#:  http:\/\/mdsst.ru\/
##CLI.ICY7#:  channels=2;samplerate=44100;bitrate=64
##CLI.META#: Р”Р¶РѕРЅ РњРѕСЂСЂРµСЃРё - Р‘Р°Р»Р»Р°РґР° Рѕ РџСЂРµРєСЂР°СЃРЅРѕР№ Р”Р°РјРµ
##CLI.PLAYING#

I (248658) audio_player: Buffer fill 15%, 65269 // 409600 bytes
WOLFSSL:0 wolfSSL error occurred, error = -305
WOLFSSL:0 wolfSSL error occurred, error = -312
E (249958) webclient: wolfSSL_read: -312, read: error during decryption, errno:0
W (249968) webclient: No data in recv. Errno = 0, error count: 1
##CLI.STOPPED# from try restart
I (250198) vs1053: watermark: 504  1284
I (251158) webclient: SSL Cleanup 1 Socket: 58
I (251278) webclient: watermark : 300  768
I (251278) webclient: IP: 3c14d85f   ADDR:95.216.20.60
W (252108) wolfssl_mp: exceeds max bit length(2048)
W (252118) wolfssl_mp: exceeds max bit length(2048)
W (252178) wolfssl_mp: exceeds max bit length(2048)
I (255578) webclient: Set Song buffer to 400k
##CLI.ICY0#:  mdsst_ru_2
##CLI.ICY3#:  http:\/\/mdsst.ru\/
##CLI.ICY5#:  64
##CLI.ICY6#:  http:\/\/mdsst.ru\/
##CLI.ICY7#:  channels=2;samplerate=44100;bitrate=64
##CLI.META#:
##CLI.META#: Р”Р¶РѕРЅ РњРѕСЂСЂРµСЃРё - Р‘Р°Р»Р»Р°РґР° Рѕ РџСЂРµРєСЂР°СЃРЅРѕР№ Р”Р°РјРµ
WOLFSSL:0 wolfSSL error occurred, error = -305
WOLFSSL:0 wolfSSL error occurred, error = -312
E (258248) webclient: wolfSSL_read: -312, read: error during decryption, errno:0
W (258248) webclient: No data in recv. Errno = 0, error count: 1

mars16, 2022-02-01:
(21338) audio_player: Buffer fill 16%, 66277 // 409600 bytes
I (52098) audio_player: Buffer fill 21%, 89549 // 409600 bytes
I (52508) webclient: SSL Cleanup 1 Socket: 59
I (52578) vs1053: watermark: 4ec  1260
I (52638) webclient: watermark : 290  656
##CLI.NAMESET#: 12 Sovetskaya estrada
##CLI.URLSET#: https://stream.gtf.club
##CLI.PATHSET#: /stream
##CLI.PORTSET#: 1250
##CLI.OVOLSET#: -36
I (52998) webclient: IP: 6691ab3e   ADDR:62.171.145.102
##CLI.VOL#: 197
I (54418) webserver: playstationInt: 12, g_device: 5
I (57618) webclient: Set Song buffer to 400k
##CLI.ICY0#: РЎРѕРІРµС‚СЃРєР°СЏ СЌСЃС‚СЂР°РґР°
##CLI.ICY3#: http:\/\/disco-radio.ru\/
##CLI.ICY4#: Retro, 80's, 70's
##CLI.ICY5#: 256
##CLI.ICY6#: РЎРѕРІРµС‚СЃРєР°СЏ СЌСЃС‚СЂР°РґР° РІСЂРµРјРµРЅ РЅР°С€РµР№ РјРѕР»РѕРґРѕСЃС‚Рё
##CLI.ICY7#:  bitrate=256
##CLI.META#: Р’РРђ \"РЎРёРЅСЏСЏ РїС‚РёС†Р°\" вЂ“ РЇ РёРґСѓ Рє С‚РµР±Рµ РЅР°РІСЃС‚СЂРµС‡Сѓ (1983)
##CLI.PLAYING#

I (60748) audio_player: Buffer fill 15%, 65014 // 409600 bytes
I (70338) audio_player: Buffer fill 0%, 1672 // 409600 bytes

mars16, 2022-02-01:
#CLI.NAMESET#: 20 Radio-Peterburg
##CLI.URLSET#: http://www.5-tv.ru
##CLI.PATHSET#: /radio.mp3
##CLI.PORTSET#: 80
##CLI.OVOLSET#: -20
I (143178) webclient: IP: a50ec97   ADDR:151.236.80.10
##CLI.VOL#: 197
I (143268) webserver: playstationInt: 20, g_device: 12
I (143358) webclient: Header: Moved
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: https://www.5-tv.ru
##CLI.PATHSET#: /radio.mp3
##CLI.PORTSET#: 443
I (144508) webclient: watermark : 290  656
I (144508) webclient: IP: a50ec97   ADDR:151.236.80.10
I (147118) webclient: Set Song buffer to 400k
##CLI.ICY0#:  DomRadio
##CLI.ICY3#:  \/
##CLI.ICY5#:  128
##CLI.ICY7#:  samplerate=44100;channels=2;bitrate=128
WOLFSSL:0 wolfSSL error occurred, error = -305
WOLFSSL:0 wolfSSL error occurred, error = -312
E (148588) webclient: wolfSSL_read: -312, read: error during decryption, errno:0
W (148588) webclient: No data in recv. Errno = 0, error count: 1
##CLI.STOPPED# from No Data
##CLI.ICY0#: No Data
I (149748) webclient: SSL Cleanup 1 Socket: 58
I (149868) webclient: watermark : 230  560

mars16, 2022-02-01:
Again, put the version to 2.1.r2 and the problems went.
Here is the right link number 3
https://stream.pcradio.ru/mdsst_ru_2-hi

karawin, 2022-02-01:
All test done with 2.2 R1

mars16, 2022-02-01:
Perhaps I have problems caused by the provider :( Thanks for your given time!

pilnikov, 2022-02-01:
{"Name":"Paradise","URL":"stream.radioparadise.com","File":"/flac","Port":"80","ovol":"0"}
{"Name":"Sector_80","URL":"stream.sectorradio.ru","File":"/geny-flac","Port":"8000","ovol":"0"}
{"Name":"progressive","URL":"89.223.45.5","File":"/progressive-160","Port":"8000","ovol":"0"}
{"Name":"paradise aac320","URL":"stream.radioparadise.com","File":"/aac-320","Port":"80","ovol":"0"}
{"Name":"Absolute Classic Rock","URL":"icecast-beta.timlradio.co.uk","File":"/absoluteclassicrock.ogg","Port":"80","ovol":"0"}
{"Name":"secure","URL":"secure.live-streams.nl","File":"/flac.flac","Port":"80","ovol":"0"}
{"Name":"freq 3","URL":"hd.stream.frequence3.net","File":"/frequence3.flac","Port":"80","ovol":"0"}
JP, Is there any hope that any of this will work?

karawin, 2022-02-01:
It is sure for empty buffer. I cannot do anything for that.

pilnikov: These are flac, but in fact ogg in the content-type. VS1053 normally can decode ogg, but ....
I don't have included the vs1053 patch for flac nor the software decoder needed, So...
The AAC is a 320 kB/s and it seems that the esp32 network has some problem, but it works on the vs1053 only. 
The software decoding fails.

pilnikov, 2022-02-01:
https://www.vlsi.fi/en/support/software/vs10xxpatches.html
?

pilnikov, 2022-02-01:
i have a board with vs1063 but .... not playing too

pilnikov, 2022-02-01:
https://github.com/pilnikov/KaRadio32_fork/blob/master/Test/Test_vs1053/Test_vs1053.ino
these - playing the ogg

karawin, 2022-02-01:
"Five versions of the patch included. With and without LATM/LOAS parser and with and without a FLAC (Free Lossless Audio Codec) decoder (1-2 channels, up to 24 bits, up to 48kHz) decoder, and with a pitch/tempo shifter."
May be I will try the patch with the flac. 

Anyway, i added the ogg type for the vs1053 so the flac is now routed to the vs1053 even if not decoded.
Update again for the new  new new 2.2R1 ;-)
http://89.223.45.5:8000/progressive-160 now is decoded in the vs1053.

pilnikov, 2022-02-01:
do it but... no sound ;(

karawin, 2022-02-01:
http://89.223.45.5:8000/progressive-160 ?

Let me the time to add the flac version ;-)

pilnikov, 2022-02-01:
ok :)

Ant-12, 2022-02-01:
Software decoding of some HE-AACv2 streams(64kBit) doesn't work since v2.2:

##CLI.NAMESET#: 7 Morena Mulher
##CLI.URLSET#: https://morenamulher.radioca.st
##CLI.PATHSET#: /stream
##CLI.PORTSET#: 443
##CLI.OVOLSET#: 0
D (799154) webclient: Socket: 59
I (799154) webclient: IP: 5c9c32c6   ADDR:198.50.156.92
##CLI.VOL#: 254
D (799164) renderer: Renderer volume max:  65536  10000
W (799944) wolfssl_mp: exceeds max bit length(2048)
W (799954) wolfssl_mp: exceeds max bit length(2048)
W (800024) wolfssl_mp: exceeds max bit length(2048)
D (801144) webclient:
Sent: GET /stream HTTP/1.1
Host: morenamulher.radioca.st
icy-metadata: 1
User-Agent: karadio32

I (801144) webclient: Set Song buffer to 400k
I (801154) webserver: playstationInt: 7, g_device: 0
D (801154) eeprom: saveDeviceSettings
WOLFSSL:0 wolfSSL error occurred, error = -305
WOLFSSL:0 wolfSSL error occurred, error = -312
E (801984) webclient: wolfSSL_read: -312, read: error during decryption, errno:0
W (801984) webclient: No data in recv. Errno = 0, error count: 1
##CLI.STOPPED# from No Data
##CLI.ICY0#: No Data
I (803134) webclient: SSL Cleanup 1 Socket: 59
I (803254) webclient: watermark : 3d0  976
D (807254) webserver: watermark serverClientTask: e8  232

Ant-12, 2022-02-01:
Software decoding of some HE-AACv2 streams(64kBit) doesn't work since v2.2:

##CLI.NAMESET#: 8 mdsst
##CLI.URLSET#: https://stream.pcradio.ru
##CLI.PATHSET#: /mdsst_ru_2-hi
##CLI.PORTSET#: 443
##CLI.OVOLSET#: 0
##CLI.VOL#: 254
D (67854) renderer: Renderer volume max:  65536  10000
I (67854) webserver: playstationInt: 8, g_device: 0
D (67854) eeprom: saveDeviceSettings
I (68724) webclient: watermark : 364  868
D (68724) webclient: Socket: 63
I (68724) webclient: IP: a714d85f   ADDR:95.216.20.167
W (69214) wolfssl_mp: exceeds max bit length(2048)
W (69224) wolfssl_mp: exceeds max bit length(2048)
W (69284) wolfssl_mp: exceeds max bit length(2048)
D (72134) webclient:
Sent: GET /mdsst_ru_2-hi HTTP/1.1
Host: stream.pcradio.ru
icy-metadata: 1
User-Agent: karadio32

I (72134) webclient: Set Song buffer to 400k
I (72164) webclient: Header: Moved
##CLI.STOPPED# from C_HDER
D (73174) webclient: parse str str6.pcradio.ru/mdsst_ru_2-hi
Strict-Transport-Security: max-age=31536000
Access-Control-Allow-Origin: *

<html>
<head><title>302 Found</title></head>
<body>
<center><h1>302 Found</h1></center>
<hr><center>nginx</center>
</body>
</html>

D (73184) webclient: parse str url https://str6.pcradio.ru
##CLI.URLSET#: https://str6.pcradio.ru
##CLI.PATHSET#: /mdsst_ru_2-hi
##CLI.PORTSET#: 443
I (73214) webclient: SSL Cleanup 1 Socket: 63
I (73334) webclient: watermark : 364  868
D (73334) webclient: Socket: 56
I (73334) webclient: IP: 6f16d85f   ADDR:95.216.22.111
W (73794) wolfssl_mp: exceeds max bit length(2048)
W (73804) wolfssl_mp: exceeds max bit length(2048)
W (73864) wolfssl_mp: exceeds max bit length(2048)
D (76684) webserver: watermark serverClientTask: 120  288
D (76764) webclient:
Sent: GET /mdsst_ru_2-hi HTTP/1.1
Host: str6.pcradio.ru
icy-metadata: 1
User-Agent: karadio32

I (76764) webclient: Set Song buffer to 400k
D (77174) webclient: contentType: 3
##CLI.ICY0#:  mdsst_ru_2
##CLI.ICY3#:  http:\/\/mdsst.ru\/
##CLI.ICY5#:  64
##CLI.ICY6#:  http:\/\/mdsst.ru\/
##CLI.ICY7#:  channels=2;samplerate=44100;bitrate=64
D (77224) webclient: t1: 0x3ffc38f2, cstatus: 2, icyfound: 1  metad:16000 Metaint:16000

D (77224) renderer: Renderer volume:  0
D (77224) renderer: Start
D (77224) webclient: newlen: 894   len: 1440   chunked:0  pdata:3ffc36d4
I (77414) audio_player: Buffer fill 2%, 11737 // 409600 bytes
D (77514) webclient: clientReceiveCallback: pdata: 3ffc36d4, pdataend: 3ffc3c74, len: 1440
D (77514) webclient: mt len:1440, clen:1440, metad:167 ,&l:3ffc377b, l:16, inpdata:3ffc36d4, rest:1256, str: StreamTitle='';
##CLI.META#:
D (77774) webclient: clientReceiveCallback: pdata: 3ffc36d4, pdataend: 3ffc3c74, len: 1440
D (77774) webclient: mt len:1440, clen:1440, metad:1048 ,&l:3ffc3aec, l:112, inpdata:3ffc36d4, rest:279, str: StreamTitle='Р”Р¶РµРє РҐРѕР»РґРµРјР°РЅ - Р®Р¶РЅС‹Р№ РґРёСЃРєРѕРјС„РѕСЂС‚ (РїРѕР»РЅР°СЏ РїРµСЂРµРґР°С‡Р°)';
##CLI.META#: Р”Р¶РµРє РҐРѕР»РґРµРјР°РЅ - Р®Р¶РЅС‹Р№ РґРёСЃРєРѕРјС„РѕСЂС‚ (РїРѕР»РЅР°СЏ РїРµСЂРµРґР°С‡Р°)
WOLFSSL:0 wolfSSL error occurred, error = -305
WOLFSSL:0 wolfSSL error occurred, error = -312
E (77984) webclient: wolfSSL_read: -312, read: error during decryption, errno:0
W (77984) webclient: No data in recv. Errno = 0, error count: 1
##CLI.STOPPED# from No Data
D (78114) renderer: Stop
##CLI.ICY0#: No Data
I (79144) webclient: SSL Cleanup 1 Socket: 56
I (79264) webclient: watermark : 364  868

karawin, 2022-02-07:
The 2.2R1 is now able to resolve all these problems.

closed 2022-02-07 by karawin