karawin/Ka-Radio32 #255: Toolchain Setup link borken

plazmax, 2022-02-01:
Hello, can you update link in HardwareConfig.md for "Toolchain Setup" dead link : 

https://dl.espressif.com/.../esp32_win32_msys2

[image]

Ant-12, 2022-02-01:
Current Ka-Radio32 supports only IDF v3.5 compatible toolchain.
It also can be built using Cmake(there need to create couple of **CMakeLists.txt** files)

h1aji, 2022-02-06:
@Ant-12 any chance you could create a pull request?

pilnikov, 2022-02-06:
i think < 0.01%
need to make a lot of adjustments

Ant-12, 2022-02-06:
This is 0%, unless [karawin](https://github.com/karawin) wins a contest for laziness.
This job is too difficult for to [make it twise.](https://github.com/karawin/Ka-Radio32/issues/252#issuecomment-1030702222)
At this moment I can post only CMakeLists.txt for ESP-IDF v3.3.5 if someone asks.

Ant-12, 2022-02-06:
Here are [my changes](https://github.com/Ant-12/Ka-Radio32/commit/646b5b5ef5dfb54079abc6522c2b4fc9b65ea7a3) for Cmake and IDF v3.3.5
But I was forced to set "Application manager  ---> Exclude PROJECT_VER from firmware image" checkbox in Espressif IoT Development Framework Configuration to avoid compiling errors.