karawin/Ka-Radio32 #256: Problems with Playing the sender "Antenne Bayern"

JocB1, 2022-02-09:
Hi, 
generally, the project runs very reliably, only with the station "Antenne Bayern" I have since V1.9 the problem that the software crashes after 1-3 hours on this station. After that only a hardware-rest helps to restart.  
URL: https://s7-webradio.antenne.de:443/antenne/stream/mp3

I use 2x Maxim MAX98357 I2S Amplifiers in Stereo configuration.

karawin, 2022-02-11:
Hi, 
I will try it. I think that the problem may be in the chunked received stream wrongly computed in some frames.

Try the simple non https   URL:http://s7-webradio.antenne.de:80/antenne/stream/mp3

JocB1, 2022-02-12:
Hi,

 

it´s the same problem with the simple non https URL.

 

Gesendet: Freitag, 11. Februar 2022 um 15:30 Uhr

karawin, 2022-02-12:
Corrected, thanks.
You can update now.

Ant-12, 2022-02-12:
entry 0x400806d8
I (8) main: starting app_main()
I (8) main: RAM left: 4378048
E (18) partition: No MD5 found in partition table
assertion "it != NULL" failed: file "C:/msys32/home/jp/esp/esp-idf/components/app_update/esp_ota_ops.c", line 507, function: esp_ota_get_running_partition
abort() was called at PC 0x400f0a17 on core 0

ELF file SHA256: c185034d148c4940

Backtrace: 0x4008d0e3:0x3ffb2da0 0x4008d4d9:0x3ffb2dc0 0x400f0a17:0x3ffb2de0 0x4014818b:0x3ffb2e10 0x400d96b3:0x3ffb2e30 0x400d2ff4:0x3ffb2e90

Ant-12, 2022-02-12:
If you have not updated the firmware for the last hour, then the solution to the problem for me was the reflashing partitions.bin 

What does "Change from IDF 4.4" mean? Are you migrating to IDF v4.4? Why not to IDF v5?

karawin, 2022-02-12:
The only usable bin (with your correction) is from KaRadio32 on git hub or update from the "update now "button of the web or the sys.update on serial.

KaRadio32_4 is not operational and may be never will be. Memory problem
And I don' know why they changed the subtype default from 0 to 6 in idf4.4.
That's why the partition must be flashed depending of the idf release.
4.4 to be sure not to have to change all deprecated in a first time.

IDF 4.4:
 cmd.exe /C "cd /D C:\Users\jp\esp\KaRadio32_4\build && C:\Espressif\python_env\...works/esp-idf-v4.4/tools/idf_size.py C:/Users/jp/esp/KaRadio32_4/build/KaRadio32.map"Total sizes:
Used static DRAM:   81812 bytes (  98924 remain, 45.3% used)
      .data size:   43172 bytes
      .bss  size:   38640 bytes
Used static IRAM:  104463 bytes (  26609 remain, 79.7% used)
      .text size:  103436 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:  186275 bytes ( 125533 remain, 59.7% used)
      .data size:   43172 bytes
      .bss  size:   38640 bytes
      .text size:  103436 bytes
   .vectors size:    1027 bytes
Used Flash size : 1615107 bytes
      .text     : 1095863 bytes
      .rodata   :  518988 bytes
Total image size: 1910377 bytes (.bin may be padded larger)

IDF 3.3:
Total sizes:
 DRAM .data size:   17212 bytes
 DRAM .bss  size:   43416 bytes
Used static DRAM:   60628 bytes (  63952 available, 48.7% used)
Used static IRAM:   72960 bytes (  58112 available, 55.7% used)
      Flash code:  926514 bytes
    Flash rodata:  547520 bytes
Total image size:~1564206 bytes (.bin may be padded larger)

Ant-12, 2022-02-12:
Look at Pilnikov's project:
https://github.com/pilnikov/Ka_mod
I got a firmware with a size of 1624496 bytes with using IDF v5.

karawin, 2022-02-12:
Yes but it i a 
#define RELEASE "1.9"
#define REVISION "7"
Two years ago and without wolfssl.

I started from Ka_Mod and written all CMakeLists in order to retrieve the real components.
Now, working on memories optimization.

Ant-12, 2022-02-12:
Yes. I know it.
Ka_Mod also has no S/PDIF output.
Maybe there need to check SDK configuration for new unused components?

Ant-12, 2022-02-20:
Jean-Pierre, please do not ignore this:
https://github.com/karawin/KaRadio32_4/issues

Ant-12, 2022-02-20:
Jean-Pierre, here I just remind you of comments on the commits.

karawin, 2022-02-21:
Corrected now.
Closing

closed 2022-02-21 by karawin