karawin/Ka-Radio32 #262: Unknown contentType mp3 stream

mars16, 2022-04-16:
Karadio32: ESP32-wroom-32+psram+vs1053, firmware v2.3.R0
Error trying to play streams:
##CLI.NAMESET#: 16 France
##CLI.URLSET#: https://live02.rfi.fr
##CLI.PATHSET#: /rfirusse-96k.mp3
##CLI.PORTSET#: 80
##CLI.OVOLSET#: -22
I (23323321) Webclient: IP: 23464ab9   ADDR:185.74.70.35
##CLI.VOL#: 180
I (23326051) webserver: playstationInt: 16, g_device: 17
I (23326741) Webclient: Set Song buffer to 400k
I (23327041) Webclient: Header: Moved
##CLI.STOPPED# from C_HDER
##CLI.URLSET#: http://rfirusse96k.ice.infomaniak.ch
##CLI.PATHSET#: /rfirusse-96k.mp3
##CLI.PORTSET#: 80
I (23328131) Webclient: SSL Cleanup 1 Socket: 55
I (23328591) Webclient: watermark : a8  168
I (23328591) Webclient: IP: 82187db9   ADDR:185.125.24.130
##CLI.ICY0#: unknown contentType
##CLI.STOPPED# from unknown contentType
##CLI.ICY0#: rfi russe 64 retransmet Ukrainian Radio
##CLI.ICY5#: 96
##CLI.ICY7#:  channels=2;samplerate=44100;bitrate=96
I (23330571) Webclient: watermark : a8  16

https://live02.rfi.fr:80/rfirusse-96k.mp3 - error
https://radio.5-tv.ru:443/radio.mp3  - error

karawin, 2022-04-19:
Right.
The server answer is Content-Type: audio/mp3
Usually it is Content-Type: audio/mpeg

I will add it.Thanks

karawin, 2022-04-19:
Update again. It is added before the next release.

closed 2022-07-17 by karawin