karawin/Ka-Radio32 #269: 1.8 inch screen connection

aternos24h, 2022-07-02:
Hi!
I have a question for you whether this screen can be connected: https: //shopee.pl/1.8-inch-TFT-LCD-Module-LCD-Screen-Module-SPI-serial-51-drivers-4-IO-driver- TFT-Resolution-128 * 160-For-Arduino-i.592946329.14623416368? Sp_atk = 435e35b8-e24f-43b6-abf3-3847d94c5518 & xptdk = 435e35b8-e24f-43b6-abf3-3847d94c5518
Until esp32 for Karadio to work?
I have not found such a scheme somewhere on the internet

aternos24h, 2022-07-02:
[image]
it is exactly this esp32 model

aternos24h, 2022-07-02:
[image]
I followed this scheme, but it turned out to be a different display ...

karawin, 2022-07-17:
Read a bit more the document please.

closed 2022-08-06 by aternos24h