karawin/Ka-Radio32 #270: OLED display and dac is not working

JayB7076, 2022-07-12:
Hi there
My name is Jay and after seeing this esp32 Web radio project video on YouTube I decided to made one for my grandfather but it seems to not working on the hardware side of esp32 I used Esp32 30pin Wroom board and max98357a dac board and oled display 0.98 inch.
First I uploaded Bin files from esp flash downloader tool, I uploaded total 4 bin files Karadio32-Karadio32-partition-bootloader and after uploading files it's worked how it mean to be I connected to my Esp32 and opened web page and it worked web page is opened successfully but after connecting oled display and dac it seems to not work display is not showing anything and dac is not working on display side did I need to change i2s address or display library?and I added some web radio site and after playing status indicates "no data".and can you tell me how can I find port number of url.

thanks to the developer of Karadio this is a great project.

karawin, 2022-07-17:
Don't know. Is SD high?

JayB7076, 2022-07-21:
No I only connected VCC-Ground and three pins for i2s communication.

JayB7076, 2022-07-23:
I uploaded Karadio32-Karadio32-partition-bootloader-standard_adb
And I used Esp32 30pin Wroom board
In first power on display not worked than I realised that I not uploaded standard_adb
And after uploading it display and rotary encoder working normally but dac is still not working.

closed 2022-07-23 by JayB7076