karawin/Ka-Radio32 #272: LCD_SPI_ST7565_NHD_C12864 too low contrast???

jamsjamsen, 2022-07-27:
I tried to use the display LCD_SPI_ST7565_NHD_C12864, and the contrast is too weak. 
I think the value the contrast is initialized at is around "100" or below, because the display appears blank at those levels.
I was run ContrastTest.ino [u8g2] and the display was easily readable at contrast levels between 110-190, and best between 130-160. (my personal choice 135).

maybe adding a new command to serial or telnet  interface would help???
 e.g. 
 sys.lcdcontrast("x")

This is a problem known in the library:
[u8g2](https://github.com/olikraus/u8g2)
  
* ST7565 NHD C12864 (issue 186)
[olikraus](https://github.com/olikraus) commented [on 13 Feb 2017](https://github.com/karawin/Ka-Radio32/issues/new#issuecomment-279310195) • 

ToDo:
- Put u8x8.setContrast(150) into init code

karawin, 2022-08-06:
Ok i will do it soon.

Va1entin0s, 2023-02-02:
Excuse me, contrast settings are available now?
like this: "u8g2.setContrast (140);"
Hardware configuration partition? Telnet? Web interface?
I need Contrast = 140 for ST7565 NHD or Contrast = 5 for ST7565 ERC

karawin, 2023-02-10:
sys.lcdblv		: Display current backlight percent value.
sys.lcdblv("x")	: Set the backlight percent value (if the hardware provide it) (range 1 to 100)

karawin, 2023-02-10:
No contrast , sorry.