karawin/Ka-Radio32 #273: I2S Configuration (Sample Rate, etc.) -> Second ESP as Bluetooth TX

JocB1, 2022-08-23:
Hi, 

I usw I2S output a.o. with MAX98xxx I2S amplifier. That works fine. The MAX supports 8...96 kHz. 

Now I need a way to stream the output from Karadio32 to a Bluetooth Speaker. I found a Projekt with ESP32:

https://github.com/pschatzmann/ESP32-A2DP
https://github.com/pschatzmann/arduino-audio-tools

Connecting Karadio I2S to the second ESP32 (I2S to Bluetooth Transmitter) unfortunately does not work for me yet. 

In base-i2s-a2dp.ino the I2S input settings has to be configured. 

I tryed this and played a bit araound: 

  auto cfg = i2s.defaultConfig(RX_MODE);
  cfg.i2s_format = I2S_STD_FORMAT; // or try with I2S_LSB_FORMAT
  cfg.bits_per_sample = 16;
  cfg.channels = 2;
  cfg.sample_rate = 44100;
  cfg.is_master    = false;   // master, no microphone                   
  i2s.begin(cfg);

My guess is that something is not set correctly here. Therefore the question which settings Karadio32 actually uses as I2S output format. From the code I have not really figured out. 

Would be great if you could get this running, this is certainly a useful thing for other users, too. 

Jochen

JocB1, 2022-08-27:
I have measured that in at I2S 48kHz is sent. Is there a possibility to change the I2S speed to 44.1 kHz?

karawin, 2022-09-04:
cfg.sample_rate = 48000;

No?

JocB1, 2022-09-10:
Yes, 
on the Bluetooth ESP32 this would be the solution, but only for the I2S side. The problem is that A2DP transmits at 44.1 kHz and the sample rate must be the same throughout the chain (according to the author of the BT ESP32 - see also the discussion at Phil's Github). This means that Karadio32 should output the I2S data with 44.1 kHz, so that the Bluetooth ESP can read them with 44.1 kHz and pass them directly to A2DP. 

Is it possible to create an additional configuration option for Karadio32 to output 44.1 kHz?

cvoinf, 2024-10-25:
I don't have a working solution, but perhaps an approach:
https://github.com/pschatzmann/arduino-audio-tools/discussions/1748#discussioncomment-11056909