karawin/Ka-Radio32 #278: wolfssl error -323, WANT_READ

HomeACcessoryKid, 2023-01-01:
Hi, I installed KaRadio32 and really like it.

The KissFM.es stream worked for me yesterday, but today it does not.
Since the wolfSSL documentation states that error -323 is not a real error, maybe the receiving function needs a bit of tweaking to handle this situation?

The URL that was redirected to works just fine in a browser, so the server is not down.
Also, the debug version of this experiment was receiving meaningful info from the server, e.g.
`##CLI.ICY5#:  125` even though the short version did not get this far.

I can help code and debug, but first want to hear your opinion on this.

TIA,
HacK

From WolfSSL documentation:
```
WANT_READ (-323)
The WANT_READ error happens often when using non-blocking sockets,
and isn’t actually an error when using non-blocking sockets, but it is passed
up to the caller as an error. When a call to receive data from the I/O callback
would block as there isn’t data currently available to receive, the I/O callback
returns WANT_READ. The caller should wait and try receiving again later.
```
Debug log:
```
> sys.version

Release: 2.2, Revision: 1, KaRadio32
> cli.start

##CLI.NAMESET#: 1 KissFMad
##CLI.URLSET#: https://adhandler.kissfmradio.cires21.com
##CLI.PATHSET#: /get_link?url=https://bbkissfm.kissfmradio.cires21.com/bbkissfm.mp3&lang=es&gdpr=1
##CLI.PORTSET#: 443
##CLI.OVOLSET#: 0
##CLI.VOL#: 254
> ##CLI.STOPPED# from C_HDER
##CLI.URLSET#: https://bbkissfm.kissfmradio.cires21.com
##CLI.PATHSET#: /bbkissfm.mp3?wmsAuthSign=c2VydmVyX3RpbWU9MDEvMDEvMjAyMyAwMjo1NjozNSBQTSZoYXNoX3ZhbHVlPVhueWlPMnRTQVFNR3JqZWxtOU9FZ0E9PSZ2YWxpZG1pbnV0ZXM9MTQ0MCZpZD02MzQzMTI3OA==
##CLI.PORTSET#: 443
WOLFSSL:0 wolfSSL error occurred, error = -323
##CLI.STOPPED# from No Data
##CLI.ICY0#: No Data

dbg.ssl("3")

##dbg.ssl is 3#
> cli.start

##CLI.NAMESET#: 1 KissFMad
##CLI.URLSET#: https://adhandler.kissfmradio.cires21.com
##CLI.PATHSET#: /get_link?url=https://bbkissfm.kissfmradio.cires21.com/bbkissfm.mp3&lang=es&gdpr=1
##CLI.PORTSET#: 443
##CLI.OVOLSET#: 0
##CLI.VOL#: 254
WOLFSSL:2 wolfSSL Entering SSL_new
WOLFSSL:3 wolfSSL Leaving SSL_new, return 0
WOLFSSL:2 wolfSSL Entering wolfSSL_dtls_set_using_nonblock
WOLFSSL:1 wolfSSL_dtls_set_using_nonblock() is DEPRECATED for non-DTLS use.
WOLFSSL:2 wolfSSL Entering SSL_set_fd
WOLFSSL:2 wolfSSL Entering SSL_set_read_fd
WOLFSSL:3 wolfSSL Leaving SSL_set_read_fd, return 1
WOLFSSL:2 wolfSSL Entering SSL_set_write_fd
WOLFSSL:3 wolfSSL Leaving SSL_set_write_fd, return 1
WOLFSSL:2 wolfSSL Entering SSL_connect()
WOLFSSL:2 wolfSSL Entering SendTls13ClientHello
WOLFSSL:1 Adding signature algorithms extension
WOLFSSL:1 Adding supported versions extension
WOLFSSL:1 growing output buffer

[... 579 lines cut ...]
WOLFSSL:3 wolfSSL Leaving wolfSSL_read_internal(), return 1440
WOLFSSL:2 wolfSSL Entering wolfSSL_read()
WOLFSSL:2 wolfSSL Entering wolfSSL_read_internal()
WOLFSSL:2 wolfSSL Entering ReceiveData()
WOLFSSL:1 Shrinking input buffer

WOLFSSL:3 wolfSSL Leaving ReceiveData(), return 1043
WOLFSSL:3 wolfSSL Leaving wolfSSL_read_internal(), return 1043
WOLFSSL:2 wolfSSL Entering wolfSSL_read()
WOLFSSL:2 wolfSSL Entering wolfSSL_read_internal()
WOLFSSL:2 wolfSSL Entering ReceiveData()
WOLFSSL:1 Embed Receive error
WOLFSSL:1 	Would block
WOLFSSL:0 wolfSSL error occurred, error = -323
WOLFSSL:3 wolfSSL Leaving wolfSSL_read_internal(), return -323
WOLFSSL:2 wolfSSL Entering SSL_get_error
WOLFSSL:3 wolfSSL Leaving SSL_get_error, return -323
WOLFSSL:2 wolfSSL Entering ERR_error_string
WOLFSSL:2 wolfSSL Entering SSL_want_read
##CLI.STOPPED# from No Data
##CLI.ICY0#: No Data
WOLFSSL:2 wolfSSL Entering SSL_free
WOLFSSL:1 CTX ref count not 0 yet, no free
WOLFSSL:3 wolfSSL Leaving SSL_free, return 0
```

HomeACcessoryKid, 2023-01-02:
Today the stream started just fine...
Still interested to know if there is this edgecase in which there is a zero payload or whatever provokes error -323.

HomeACcessoryKid, 2023-01-10:
I feel a bit stupid, I never upgraded to 2.3R0!

With that new version the problem is fixed.
And it even says so in the release notes!
`Bugs correction: HTML chunks processing and Meta computing`

One request though: please provide 2.3R0 binaries in the Karadio32 repo on GitHub?
Because now that is 2.2R1, I thought 2.3R0 was not production ready...

karawin, 2023-01-15:
Done

HomeACcessoryKid, 2023-01-15:
Maybe something went wrong, but I still see release 2.2R1 as the latest release on GitHub...
Can you check again?  Thanks!

karawin, 2023-01-15:
Yes, i forgot to sync the view.
Done now, thanks.

HomeACcessoryKid, 2023-01-15:
Maybe we have an issue with expectations:
I currently see 2.2R1 as the latest release of github.com/karawin/Ka-Radio32
I expected that to become 2.3R0

And maybe also to see 2.3R0 on github.com/karawin/Ka-Radio32_4 which currently has no releases