karawin/Ka-Radio32 #281: Does not get the ip address

ewgenw, 2023-02-17:
karadio32 on esp32-wroom board does not get ip address from dhcp . in telnet i get last message "E (4010) event: invalid static ip" and it's all. 
On my wifi access point i see connected board with address 0.0.0.0 . 
When i write static ip - board write in telnet that address is which I entered, but there is no connection , and acess point show address 0.0.0.0

karawin, 2023-03-08:
Please read the documentation.
The first time KaRadio starts in AP mode to be able to configure the wifi.

ewgenw, 2023-03-08:
Yes , i did it - in AP mode I have configured the name and password of my AP. After rebooting, the board connects to the AP, but does not receive an IP address.
this is how it looks on my AP
[image]

And this is console of ESP board
[image]

karawin, 2023-03-08:
DHCP checked on the interface?