karawin/Ka-Radio32 #282: Statio URL Convert

Beppi4U, 2023-03-01:
Hi! 
Buil one with VS1053 and 1.3 OLED 
Workes like a charme from the start!
But when i try to put in my favourite stream URL's i only get errors!
How to put in these URLs correct?
Thank you and Greetings
Andi

LIST:  Ignore Presets  and after #
preset_00 = swr-edge-20b5-fra-lg-cdn.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3 # 0 - SWR3
preset_01 = swr-edge-10b5-fra-dtag-cdn.cast.addradio.de/swr/swr/raka05/mp3/128/stream.mp3 #  1 - SWR3 ROCK
preset_02 = 185.52.127.159/de/56855/mp3_128.mp3?origine=ecoute_de #  2 - NRJ RADIO
preset_03 = 22333.live.streamtheworld.com/AFNE_STU_SC?csegid=12 # 3 - AFN The Eagle (Stuttgart)
preset_04 = stream.antenne1.de/stream1/livestream.mp3 (128)# 4 - ANTENNE1
preset_05 = streams.bigfm.de/bigfm-deutschland-128-aac #  5 - BIG FM RADIO
preset_06 = edge.live.mp3.mdn.newmedia.nacamar.net/ps-dieneue_rock/livestream_hi.mp3 #  6 - Die Neue 107.7
preset_07 = mp3tx.duplexfx.com:8800 #  7 - ENERGY98 Dance Music
preset_08 = webcast.wonc.org:8000/wonclive-128s #  8 - Pure Rock FM 89 USA
preset_09 = stream.inselradio.com:20800/high.inselradio.mp3 #  9 - INSEL RADIO MALLORCA
preset_10 = stream.inselradio.com/high.sunshine.mp3 # 10 - SUNSHINE MALLORCA
preset_11 = mp3-live.swr.de/swr4bw_m.m3u # 11 - SWR4 SCHLAGERALARM!
preset_12 = swr-swr1-bw.cast.addradio.de/swr/swr1/bw/mp3/128/stream.mp3 # 12 - SWR1 BW
preset_13 = swr-dasding-live.cast.addradio.de/swr/dasding/live/mp3/128/stream.mp3 # 13 - DAS DING LIVE
preset_14 = str2.creacast.com/sudtirol1a # 14 - RADIO SUEDTIROL
preset_15 = uk3.internet-radio.com:8405/live # 15 - Majestic Jukebox Radio
preset_16 = listen.radiomax.technology/penny # 16 - Penny Live Radio
preset_17 = dradio-edge-30af-fra-ts-cdn.cast.addradio.de/dradio/dlf/live/mp3/128/stream.mp3 # 17 - Deutschlandfunk
preset_18 = streams.regenbogen.de/rr2-rn-128-aac # 18 - Radio Regenbogen2

karawin, 2023-03-08:
Not a KaRadio software  Issue.......
Try [Edzelf] project.

Beppi4U, 2023-03-08:
These are just example streams that worked with EdZelf's WebRadio.
Thought i could easy copy&paste ......but no!! ;-)
Well at least, after a few try&error, i figured it out!!
Will post the list after collecting my favourite stations!
How to avoid the display from going to showing clock after a while?
I want to see the station names continously!
Thanx for helping out!
Andi

[WebStations(1).txt](https://github.com/karawin/Ka-Radio32/files/10924849/WebStations.1.txt)

karawin, 2023-03-08:
Double click on the encoder to toggle the display mode.

Beppi4U, 2023-03-09:
That's right! 
Merci !

closed 2023-03-09 by Beppi4U