karawin/Ka-Radio32 #284: Wrong play aac

electro-nic, 2023-03-13:
Hi! I add new station http://radio-srv1.11one.ru:80/rec64k to playlist and it not play
[image]

what logs can i help?
regards

pilnikov, 2023-03-14:
[image]
[image]

electro-nic, 2023-03-14:
not play music but title is ok. After switch to another station esp freeezing
```

##CLI.LIST#
##CLI.INFO#
##SYS.DATE#: 2023-03-14T19:59:22+02:00
##CLI.NAMESET#: 223 S7
##CLI.ICY0#: Ready
##CLI.VOL#: 19
##CLI.STOPPED# from State
##CLI.NAMESET#: 223 S7
##CLI.URLSET#: http://radio-srv1.11one.ru
##CLI.PATHSET#: /rec64k
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 64
##CLI.VOL#: 19
##CLI.ICY0#: http:\/\/radiorecord.fm
##CLI.ICY3#: http:\/\/radiorecord.fm
##CLI.ICY4#: Dance
##CLI.ICY5#: 64
##CLI.ICY6#: Radio Record 64k AAC+
##CLI.ICY7#:  channels=2;samplerate=44100;bitrate=64
##CLI.META#: JOEL CORRY\/TOM GRENNAN - Lionheart (Fearless) @ radiorecord.fm
##CLI.PLAYING#
```

[https://youtube.com/shorts/hqyaEWoGXNw?feature=share](url)

pilnikov, 2023-03-15:
а до этого воспроизводились ААС станции?
а звуковой чип VS1053?
а начало лога можно посмотреть?
Have you played AAC stations before?
What about the VS1053 sound chip?
can you see the beginning of the log?

electro-nic, 2023-03-15:
I  don't know. I recently just to try play. 

```
##CLI.INFO#
##SYS.DATE#: 2023-03-15T11:29:42+02:00
##CLI.NAMESET#: 0 RadioROKS HardnHeavy
##CLI.ICY0#: Radio ROKS Hard'n'Heavy
##CLI.ICY3#: https:\/\/www.radioroks.ua
##CLI.ICY4#: Hard'n'Heavy
##CLI.ICY5#: 320
##CLI.ICY6#: Only Hard'n'Heavy!
##CLI.META#: Edguy - Out Of Control
##CLI.VOL#: 57
##CLI.PLAYING#
##CLI.NAMESET#: 1 Radio ROKS ��������
##CLI.URLSET#: https://online.radioroks.ua
##CLI.PATHSET#: /RadioROKS_HD
##CLI.PORTSET#: 443
##CLI.OVOLSET#: 63
##CLI.VOL#: 57
##CLI.ICY0#: Radio ROKS
##CLI.ICY3#: https:\/\/www.radioroks.ua
##CLI.ICY4#: Rock
##CLI.ICY5#: 320
##CLI.ICY6#: Radio ROKS
##CLI.META#: 
##CLI.PLAYING#
##CLI.NAMESET#: 223 S7
##CLI.URLSET#: http://radio-srv1.11one.ru
##CLI.PATHSET#: /rec64k
##CLI.PORTSET#: 80
##CLI.OVOLSET#: 64
##CLI.VOL#: 57
##CLI.ICY0#: http:\/\/radiorecord.fm
##CLI.ICY3#: http:\/\/radiorecord.fm
##CLI.ICY4#: Dance
##CLI.ICY5#: 64
##CLI.ICY6#: Radio Record 64k AAC+
##CLI.ICY7#:  channels=2;samplerate=44100;bitrate=64
##CLI.META#: Record Dance Radio @ radiorecord.fm
##CLI.PLAYING#
##CLI.META#: KILLTEQ\/D.HASH\/VALLHEE - I Like It @ radiorecord.fm
```
maybe I can make a more detailed debug?

pilnikov, 2023-03-15:
sys.logv
and view log from start

pilnikov, 2023-03-15:
15:25:24.250 -> entry 0x400805ec
15:25:25.711 -> E (1485) main: RAM left: 4433105
15:25:25.711 -> E (1495) main: Running partition type 0 subtype 16 (offset 0x00010000)
15:25:25.997 -> Log level is now ESP_LOG_INFO
15:25:26.138 -> I (1885) vs1053: VS1053 LFreq: 1403508
15:25:26.138 -> I (1885) vs1053: VS1053 HFreq: 6153846
15:25:26.750 -> I (2485) vs1053: SPI_AUDATA 1 = 1f40
15:25:26.750 -> I (2485) vs1053: VS10xx detection. Version: 4
15:25:26.750 -> I (2485) VS1053 Patches: plugin size 8414, start: 7 1 8050
15:25:27.314 -> I (3055) VS1053 Patches: plugin end Addr: a,  val:0 50
15:25:27.314 -> I (3055) vs1053: SPI_AUDATA 2 = 1f40
15:25:27.314 -> I (3065) vs1053: SCI_Status  = 0x40
15:25:27.314 -> I (3065) vs1053: SCI_Mode (0x4800) = 0x4802
15:25:27.314 -> I (3065) vs1053: SCI_ClockF = 0xB800
15:25:27.314 -> I (3065) vs1053: VS Version (VS1053 is 4) = 4
15:25:27.314 -> I (3075) vs1053: I2S Speed: 0
15:25:27.314 -> I (3085) vs1053: volume: 93
15:25:27.314 -> I (3085) main: hardware initialized
15:25:27.360 -> I (3115) main: audio_output_mode 4

as an example

pilnikov, 2023-03-15:
and

15:25:27.360 -> I (3115) main: Project name: KaRadio32_4
15:25:27.360 -> I (3115) main: Version: 2.3.0
15:25:27.360 -> I (3125) main: Release 2.3, Revision 0
15:25:27.360 -> I (3125) main: SDK v5.0-dev-3654-gc2ccc383da-dirty
15:25:27.360 -> I (3125) main:  Date Jul  6 2022, Time: 11:56:34
15:25:27.360 -> 
15:25:27.360 -> I (3135) main: Heap size: 4429401

electro-nic, 2023-03-15:
```
Using '/dev/ttyUSB0' as serial port.
Showing logs:
[16:27:41]3epom: saveDeveetns
[16:27:41]39525) erLog sow EOERBOSE
[16:27:41]D (icStigi nG_(339615) aIN  EPLGBOE
[16:27:41]D (icStigi nG_(339615) aIN  EPLGBOE
[16:27:41]level SP_LV Lew_VR
[16:28:09]Vddon: E:Log lvel is noS_OES368195EN8
[16:28:09]Vddon: E:Log lvel is noS_OES368195EN8
[16:28:09] () TLET: ##SYS.HEAPd0: 389470 #
[16:28:09]ys.bt
[16:28:09]VV (368195) TELNET: brec: soo
[16:28:09]ys.bt
[16:28:09]e: r)
[16:28:09]t(0I (368tme: 19357204399277 us
[16:28:09]I (368195) wifi:staun -> init 195) wifi: usold:<6,0>, a<255,255>, sta:<6,0>, prof:1
[16:28:09]t(0I (368tme: 19357204399277 us
[16:28:09]pm stop, toal sleep ti / 3630
[16:28:09] (5) main: WifiDisnected.
[16:28:09] (5) main: WifiDisnected.
Serial port closed!
[16:28:09]3I(835) wif:flus txq
[16:28:09]62iptxq
[16:28:09] 362ihI (3835) wif:sto sw I (368235) wifi:a stop hw txq
[16:28:09]lmces Jun  8 21 02:7
[16:28:09]lmces Jun  8 21 02:7
[16:28:09]t060:25
[16:28:09]:0xcPUSET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:28:09]:0xcPUSET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:28:09]rst (SW_C_REconfigsip: 0, SPIWP:0xee
[16:28:09]v:x00q_drv:0x00,d_dr:0x00,sdrv0x,:0x0
[16:28:09]clk_dr0,vc0_:00hd_drv00,wp_drv:0x0mode:DOiv:2
[16:28:09]
[16:28:09]I, clock dload:0x3fff0018,len:4load:0x3fff001c,len:4532
[16:28:09]:0x4007800len:24
[16:28:09]o 0tail 12 room 4
[16:28:09]ad:0x40080400,len:00
[16:28:09]
[16:28:11](25) main: startig ap_main()
[16:28:11],tenal 217036
[16:28:11]load0,107h lo56entry 0x400806b4I 1npI (125) main: RAM left: 4394956 InrE3)mi: ubtof0x000100)
[16:28:11] (15 an Running partition type 0sype 16 (fset 00I(155) ma: Pt done...
[16:28:11] inartition iniDIO: gpio_get_pinDeepSleep err 0x1102
[16:28:11] inartition iniDIO: gpio_get_pinDeepSleep err 0x1102
[16:28:11] (245) GPD (245) eeprom: saveDeviceSettings
[16:28:11]og level is now ESP_LOG_VERBOSE
[16:28:11]LV (335) addo LINEog leveRBS
[16:28:11]n:: Ll is now ESP_LOG_VEOE
[16:28:11]LV (335) addo LINEog leveRBS
[16:28:11]I (425) vs1053: VS1053 LFreq: 1600000
[16:28:11]I (425) vs1053: VS1053 LFreq: 1600000
[16:28:11]I (425) vs1053: VS1053 HFreq: 6153846
[16:28:12] (1025) SPUD= 1f40
[16:28:12]I vs1053:I_AATA 1 I (1025) vs1053: Vxx detection. Version: 4
[16:28:12], start:   00
[16:28:12]S10I (1025) VS1053 Patches: plugin size 84147185I(1535 VS1053 Patches: plugin end Addr: a,  val: 50
[16:28:12]S10I (1025) VS1053 Patches: plugin size 84147185I(1535 VS1053 Patches: plugin end Addr: a,  val: 50
[16:28:12] )0I (1535) vs1053: SPI_AUDATA 2 = 1f40
[16:28:12]I 35) vs153CI_tatus  = 0x40
[16:28:12]0x800) = 0x4802
[16:28:12]153: SCI_ClockF = 0xB800
[16:28:12](150: SSI (1535) vs1053: SCI_Mode (4I (5) vs1053I (1545) vs13S rson(S53 s 4)  4I (1545) vs1053: I2S Speed: 0
[16:28:12](150: SSI (1535) vs1053: SCI_Mode (4I (5) vs1053I (1545) vs13S rson(S53 s 4)  4I (1545) vs1053: I2S Speed: 0
[16:28:12]05: VVei V10i=
[16:28:12]v5:vlm:0
[16:28:12]I (1555) s103 oue I (1555) mainhrwe inid
[16:28:12] 57)GI:gtepa r x12
[16:28:12]: adartializeD(15 PO e_slyer010I (1575) mao_oe4
[16:28:12]MEU,DAC_UL_N D,V15,SDF
[... 772 lines cut ...]
[16:29:57]V (106325) websocket: websocketbroadcast: {"wsicy":{"curst":"223","descr":"","meta":"","name":"","bitr":"","url1":"","not1":"","not2":"","genre":""}}
[16:29:57]V (106345) Webclient: Client incfree of 3ffc0f48, from wsh           Heap size: 3881960
[16:29:59]E (107885) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[16:29:59]E (107885) task_wdt:  - IDLE1 (CPU 1)
[16:29:59]E (107885) task_wdt: Tasks currently running:
[16:29:59]E (107885) task_wdt: CPU 0: clientTask
[16:29:59]E (107885) task_wdt: CPU 1: vsTask
[16:29:59]E (107885) task_wdt: Print CPU 0 (current core) backtrace
[16:29:59]
[16:29:59]
[16:29:59]Backtrace:0x400D3375:0x3FFB06B0 0x40083DED:0x3FFB06D0 0x4008EB97:0x3FFD9F30 0x4008EB92:0x3FFD9F50 0x400DE8E4:0x3FFD9FB0 0x400DEC6F:0x3FFD9FD0 0x400E126A:0x3FFDA020 
[16:29:59]
[16:29:59]E (107885) task_wdt: Print CPU 1 backtrace
[16:29:59]
[16:29:59]
[16:29:59]Backtrace:0x40083853:0x3FFB0CB0 0x40083DED:0x3FFB0CD0 0x4008EB50:0x3FFC02D0 0x400DE54D:0x3FFC02F0 0x400DE8E4:0x3FFC0350 0x400DED03:0x3FFC0370 
[16:29:59]
[16:30:04]E (112885) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[16:30:04]E (112885) task_wdt:  - IDLE1 (CPU 1)
[16:30:04]E (112885) task_wdt: Tasks currently running:
[16:30:04]E (112885) task_wdt: CPU 0: clientTask
[16:30:04]E (112885) task_wdt: CPU 1: vsTask
[16:30:04]E (112885) task_wdt: Print CPU 0 (current core) backtrace

```

pilnikov, 2023-03-15:
Я могу только предложить свой бинарник (он собран в более свежей версии sdk и использует более свежие патчи для vs1053) 
Судя по вашим логам затык проходит на этапе декодирования

16:29:39]E (87885) task_wdt: CPU 1: vsTask
[16:29:39]E (87885) task_wdt: Print CPU 0 (current core) backtrace
[16:29:39]
[16:29:39]
[16:29:39]Backtrace:0x400D3375:0x3FFB06B0 0x40083DED:0x3FFB06D0 0x401ADFDF:0x3FFC1300 0x400D459B:0x3FFC1320 0x4008D335:0x3FFC1340 

Подвисает vsTask, а это как раз таки аппаратный декодер.
Для vs1053 производитель иногда выпускает патчи(заплатки), которые исправляют косяки в декодере. Возможно у меня они свежее.

electro-nic, 2023-03-15:
Please close issue. This hardware problem (vs1053b want normal voltage CVDD).

pilnikov, 2023-03-16:
[KaRadio32_4.zip](https://github.com/karawin/Ka-Radio32/files/10986220/KaRadio32_4.zip)
try and close

electro-nic, 2023-03-19:
this is hardware problem

closed 2023-03-19 by electro-nic